riscv-software-src/riscv-tests #303: Invalid or unknown z ISA extension: 'zfh'

bitglitcher, 2020-11-20:
I get this issue when trying to compile. I followed all the steps on the README. 
It might be my RISC-V toolchain installation. 
This is my install path. `/home/user/Documents/opt/riscv`
I have `riscv64-unknown-elf-gcc` with `multilib` support.

This is my RISC-V GCC Version.

```
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/mnt/c/Users/camin/Documents/opt/riscv/libexec/gcc/riscv64-unknown-elf/10.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /mnt/c/Users/camin/Documents/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-elf --prefix=/mnt/c/Users/camin/Documents/opt/riscv --disable-shared --disable-threads --enable-languages=c,c++ --with-system-zlib --enable-tls --with-newlib --with-sysroot=/mnt/c/Users/camin/Documents/opt/riscv/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././riscv-gcc --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket 'CFLAGS_FOR_TARGET=-Os   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os   -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 10.2.0 (GCC)
```

I get this error when trying to compile the RISCV tests. I get multiple of these errors.
```
Fatal error: -march=rv32imafd_zfh: Invalid or unknown z ISA extension: 'zfh'
```

aswaterman, 2020-11-21:
I think you either need a more recent version of binutils that supports the Zfh extension, or you need to revert to an earlier commit of riscv-tests, e.g. c4217d88bce9f805a81f42e86ff56ed363931d69

aswaterman, 2020-11-21:
I've made a patch that will only attempt to build or run the tests supported by your installed compiler.

closed 2020-11-21 by aswaterman

bitglitcher, 2020-11-21:
> I've made a patch that will only attempt to build or run the tests supported by your installed compiler.

Thanks, now it works fine.

commit 58eb560a (2020-11-21): Only attempt to build tests supported by compiler

Resolves #303