riscv-software-src/riscv-tests #304: unknown z ISA extension: 'zfh'

zhuzhzh, 2020-11-21:
I am using riscv64-unknow-elf-gcc 10.2.0. 
With it, the compiler will fail due to unknown z ISA extension: 'zfh'.
Does anyone meet the same issue?

```
Assembler messages:
Fatal error: -march=rv32imafd_zfh: Invalid or unknown z ISA extension: 'zfh'
make[1]: *** [/trunk/branch/edahub/riscv-tests/build/../isa/Makefile:82: rv32uzfh-p-fadd] Error 1
make[1]: Leaving directory '/trunk/branch/edahub/riscv-tests/build/isa'
make: *** [Makefile:29: isa] Error 2

# harriszh @ pc-harriszh2 in /trunk/branch/edahub/riscv-tests/build on git:master o [18:51:54] C:2
$ vi ../isa/Makefile 

# harriszh @ pc-harriszh2 in /trunk/branch/edahub/riscv-tests/build on git:master o [18:52:57] 
$ riscv64-unknown-elf-gcc --version
riscv64-unknown-elf-gcc (GCC) 10.2.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

pavelkryukov, 2020-11-21:
> Does anyone meet the same issue?

Yes, it is reported here: https://github.com/riscv/riscv-tests/issues/303

zhuzhzh, 2020-11-22:
@pavelkryukov Thanks, I roll back the old commit. It doesn't have such issue.