riscv-software-src/riscv-tests #305: how to run dhrystone with spike?

zhuzhzh, 2020-11-22:
I run it with the command `spike pk benchmarks/dhrystone.riscv`

```bash
$ spike pk benchmarks/dhrystone.riscv
bbl loader
couldn't open ELF program: benchmarks/dhrystone.riscv!
```

I checked the compile command. (riscv64-unknown-elf-gcc v10.2.0)

```bash
$ riscv64-unknown-elf-gcc-10.2.0 -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc-10.2.0
COLLECT_LTO_WRAPPER=/home/harriszh/.local/riscvtool/libexec/gcc/riscv64-unknown-elf/10.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /trunk/branch/edahub/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-elf --prefix=/home/harriszh/.local/riscvtool --disable-shared --disable-threads --enable-languages=c,c++ --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/harriszh/.local/riscvtool/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././riscv-gcc --disable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket 'CFLAGS_FOR_TARGET=-Os   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os   -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 10.2.0 (GCC)
```

```txt
riscv64-unknown-elf-gcc -I/trunk/branch/riscv/riscv-tools/riscv-tests/build/../benchmarks/../env -I/trunk/branch/riscv/riscv-tools/riscv-tests/build/../benchmarks/common -I/trunk/branch/riscv/riscv-tools/riscv-tests/build/../benchmarks/median -I/trunk/branch/riscv/riscv-tools/riscv-tests/build/../benchmarks/qsort -I/trunk/branch/riscv/riscv-tools/riscv-tests/build/../benchmarks/rsort -I/trunk/branch/riscv/riscv-tools/riscv-tests/build/../benchmarks/towers -I/trunk/branch/riscv/riscv-tools/riscv-tests/build/../benchmarks/vvadd -I/trunk/branch/riscv/riscv-tools/riscv-tests/build/../benchmarks/multiply -I/trunk/branch/riscv/riscv-tools/riscv-tests/build/../benchmarks/mm -I/trunk/branch/riscv/riscv-tools/riscv-tests/build/../benchmarks/dhrystone -I/trunk/branch/riscv/riscv-tools/riscv-tests/build/../benchmarks/spmv -I/trunk/branch/riscv/riscv-tools/riscv-tests/build/../benchmarks/mt-vvadd -I/trunk/branch/riscv/riscv-tools/riscv-tests/build/../benchmarks/mt-matmul -I/trunk/branch/riscv/riscv-tools/riscv-tests/build/../benchmarks/pmp -DPREALLOCATE=1 -mcmodel=medany -static -std=gnu99 -O2 -ffast-math -fno-common -fno-builtin-printf -o dhrystone.riscv /trunk/branch/riscv/riscv-tools/riscv-tests/build/../benchmarks/dhrystone/dhrystone.c /trunk/branch/riscv/riscv-tools/riscv-tests/build/../benchmarks/dhrystone/dhrystone_main.c /trunk/branch/riscv/riscv-tools/riscv-tests/build/../benchmarks/common/syscalls.c /trunk/branch/riscv/riscv-tools/riscv-tests/build/../benchmarks/common/crt.S -static -nostdlib -nostartfiles -lm -lgcc -T /trunk/branch/riscv/riscv-tools/riscv-tests/build/../benchmarks/common/test.ld
```

I opened dhrystone.riscv.dump. I didn't find any issue.

```
dhrystone.riscv:     file format elf64-littleriscv
Disassembly of section .text:

0000000080001048 <Proc_7>:
    80001048:   2509                    addiw   a0,a0,2
    8000104a:   9da9                    addw    a1,a1,a0
    8000104c:   c20c                    sw  a1,0(a2)
    8000104e:   8082                    ret
```

I think I made some mistake there. Could anyone show the right way to run it?

iamywang, 2021-02-04:
just like this:
```
$ spike benchmarks/dhrystone.riscv
```

[image]

iamywang, 2021-02-04:
Just this is ok!
```
$ spike benchmarks/dhrystone.riscv
```

[image]

zhuzhzh, 2021-02-25:
it doesn't work on my side.

```
$ ~/.local/riscv/bin/spike benchmarks/dhrystone.riscv
terminate called after throwing an instance of 'std::runtime_error'
  what():  bad syscall #2147504448
```

TerrenceSun, 2021-06-04:
Try the gcc option mentioned in https://github.com/riscv/riscv-pk/issues/170, it's very likely the same issue.

sazczmh, 2021-08-29:
> Just this is ok!
> 
> ```
> $ spike benchmarks/dhrystone.riscv
> ```
> 
> [image]

Why dhrystone does not need pk?

thanks.

TerrenceSun, 2021-08-29:
> > Just this is ok!
> > ```
> > $ spike benchmarks/dhrystone.riscv
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > [image]
> 
> Why dhrystone does not need pk?
> 
> thanks.

dhrystone starts at ddr boot address 0x80000000. It is a standalone bare metal program, it has its own syscall implementation.  This is also why it has the same memset issue (bad syscall message) as pk.

sazczmh, 2021-08-29:
> dhrystone starts at ddr boot address 0x80000000. It is a standalone bare metal program, it has its own syscall implementation. This is also why it has the same memset issue (bad syscall message) as [...]
It's clear to me now. Thank you very much.

Dolu1990, 2021-09-08:
I'm curious, why the number of instruction per dhrystone run is 392 ?
Should it not be lower ? around ~323 (i don't remember exactly).

(i just compiled and had the same results than you)

Dolu1990, 2021-09-08:
If i'm right, 392 instruction per run, on a single issue CPU running full speed would mean 1.45 dmips/mhz  (1000000÷1757÷392), which isn't right for RISC-V.

GGGjiji, 2022-08-22:
> it doesn't work on my side.
> 
> ```
> $ ~/.local/riscv/bin/spike benchmarks/dhrystone.riscv
> terminate called after throwing an instance of 'std::runtime_error'
>   what():  bad syscall #2147504448
> ```

I also encountered the same situation.
My analysis is that the dhrystone.riscv has some problems. So I change the compile option -o2 to -o1 in file /riscv-tests/benchmarks/Makefile and re-make.
In my environment this trick worked.Hope it works as well for you