riscv-software-src/riscv-tests #312: BSS reset problem in pmp.riscv test

Rolfy360, 2020-12-18:
In pmp tests,
these 2 arrays are not set to zero if there is no BSS set-to-zero.
static uintptr_t counters[NUM_COUNTERS];
static char* counter_names[NUM_COUNTERS];

The _init() will try to print strings from counter_names[].
If the memory reset value of the platform are not zero,
the counter_names[] are wild pointers.

Same problem happens in rv*-v-* tests.
The page table used by vm.c is in the BSS section.
For C code, the BSS section should be set to zero.

Rolfy360, 2020-12-18:
I insert the code below in crt.S, pmp can pass in my platform.
============================
init_bss:
  la a0, __bss_start ## need add this in test.ld
  la a2, _tdata_begin ## the tdata/tbss are handled by init_tls()
  sub a2, a2, a0
  li a1, 0
  jal memset

  li a0, 0
  li a1, 0
  j _init

aswaterman, 2020-12-19:
Different platforms have different conventions about who is responsible for zeroing the bss; in many contexts, it's the loader's responsibility, and in a handful of others, it's the crt's responsibility.  These programs assume it's the loader's responsibility.

closed 2021-06-09 by Rolfy360