riscv-software-src/riscv-tests #315: rv32ua Atomic LR/SC test 4 is overly constrained.

ben-marshall, 2021-01-03:
Hi

I think test 4 of the `rv32ua` (and `rv64ua`) tests is wrong. The [ISA Spec](https://github.com/riscv/riscv-isa-manual/releases/download/draft-20201229-5890a1a/riscv-spec.pdf) (Section 8.2) says that

> An implementation can register an arbitrarily large reservation set on each LR, provided the reser-
vation set includes all bytes of the addressed data word or doubleword.

Which means that, theoretically, an implementation could place a reservation on the entire address space, right?

Whereas [this test](https://github.com/riscv/riscv-tests/blob/7d069666a5841e5e5b3ba1723c6af26925a35a9c/isa/rv64ua/lrsc.S#L41) suggests that only an LR/SC to the _same address_ can be successful. Either that, or the test requires an implementation which registers only some maximum size of reservation set. This seems at odds with the spec.

I'm happy to submit a patch which removes the test, but I wanted to get another reading from other people.

Cheers,
Ben

aswaterman, 2021-01-04:
Yes, this test was written long ago and implicitly assumes the reservation granule is < 1 KiB.

If supervisor mode is present, there is a 4 KiB upper bound, but this test isn't necessarily run in that environment.  So I think the most direct resolution would be to remove the test for now.

aswaterman, 2021-01-04:
Resolved via #316

closed 2021-01-04 by aswaterman