riscv-software-src/riscv-tests #321: Errorneously recursive code generated for non-recursive memset() in syscalls.c with -O2

quxfred, 2021-01-30:
https://github.com/riscv/riscv-tests/blob/e75931826097c741f274212a49302e39ca597168/benchmarks/Makefile#L40

[image]

Note the recursion marked in orange in above disassembly is supposed to perform the else-branch at common/syscalls.c:406. But this just causes a stack overflow. Adding  -fno-tree-loop-distribute-patterns or -ffreestanding got rid of this error. It looks having to do with a recurring GCC bug that attempts some obscure transformation on memset-like code.

See also:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=56888
https://gcc.gnu.org/legacy-ml/gcc-help/2019-10/msg00060.html
https://gcc.gnu.org/legacy-ml/gcc-help/2019-10/msg00076.html

Regards,
Freddie

aswaterman, 2021-01-30:
See
https://github.com/riscv/riscv-gcc/issues/197

quxfred, 2021-01-30:
So this has been known for months. I'll close this issue then.

Should the work-around be in the Makefile?

aswaterman, 2021-01-30:
It should be in the Makefile, but someone needs to take the time to figure out how to do it without being too disruptive to performance. -fno-builtin is too heavy of a hammer because it destroys dhrystone performance for other reasons. Jim’s suggestion on the ticket I linked to is probably the right answer, but that needs to be validated, by making sure that dhrystone’s use of string copy, struct copy, etc. are still inlined.

(Note the reason it hasn’t been fixed yet is that older versions of the compiler don’t exhibit this problem. Not that that fact helps with your situation, I realize.)

aswaterman, 2021-01-30:
I’ll reopen the ticket since the problem isn’t actually fixed.

quxfred, 2021-01-31:
Just made a quick-and-dirty patch in PR #322. Sorry I don't know how to put it under this issue.

aswaterman, 2021-02-02:
I double-checked that the solution proposed in https://github.com/riscv/riscv-gcc/issues/197 works on older compilers, too, and it's a cleaner fix.  I'll commit that and close this issue.

commit ea85805f (2021-02-02): Prevent GCC from pattern-matching the memset implementation

Otherwise, we get infinite recursion.

Resolves #321
Resolves #322
closed 2021-02-02 by aswaterman