riscv-software-src/riscv-tests #328: the problem about rv32ui-p-sh

DTongTong, 2021-02-20:
Hi, I met some  problem running the rv32ui-p-sh. I manually calculated the test_4 code segment process in the program and found that it always jumped to the fail code segment. The following is the corresponding disassembly code ：
000001c0 <test_4>:
 1c0:	        00001097          	auipc	ra,0x1
 1c4:         48008093          	addi	ra,ra,1152 # 1640 <__DATA_BEGIN__>
 1c8: 	beef1137          	lui	sp,0xbeef1
 1cc:	        aa010113          	addi	sp,sp,-1376 # beef0aa0 <__global_pointer$+0xbeeeec60>
 1d0:	00209223          	sh	sp,4(ra)
 1d4:	0040af03          	lw	t5,4(ra)
 1d8:	beef1eb7          	lui	t4,0xbeef1
 1dc:	aa0e8e93          	addi	t4,t4,-1376 # beef0aa0 <__global_pointer$+0xbeeeec60>
 1e0:	00400193          	li	gp,4
 1e4:	41df1663          	bne	t5,t4,5f0 <fail>
I think in the address 1d0 sh should be sw.
The same problem is also met in the rv32ui-p-sb program.
Could you please have a look and tell me the solution?
Thanks very much!

Derrick0415, 2023-03-15:
Just ran into the same problem, I think TEST_4 is wrong

aswaterman, 2023-03-15:
The test is valid, despite its construction being perhaps unexpected.

closed 2023-03-15 by aswaterman