riscv-software-src/riscv-tests #330: Problems when running executables in gem5

minhoha, 2021-03-10:
When I run executable file, like _dhrystone.riscv_, in gem5 SE mode, core dump error occurs like below.

> warn: CheckedInt already exists in allParams. This may be caused by the Python 2.7 compatibility layer.
> warn: Enum already exists in allParams. This may be caused by the Python 2.7 compatibility layer.
> warn: ScopedEnum already exists in allParams. This may be caused by the Python 2.7 compatibility layer.
> gem5 Simulator System.  http://gem5.org
> gem5 is copyrighted software; use the --copyright option for details.
> 
> gem5 version 20.0.0.0
> gem5 compiled Jan  5 2021 05:22:07
> gem5 started Mar 10 2021 17:55:46
> gem5 executing on rpm-System-Product-Name, pid 9446
> command line: ./build/RISCV/gem5.opt ./configs/example/se.py --cmd /home/mhha/riscv-tests/benchmarks/dhrystone.riscv
> 
> Global frequency set at 1000000000000 ticks per second
> warn: DRAM device capacity (8192 Mbytes) does not match the address range assigned (512 Mbytes)
> warn: Unknown operating system; assuming Linux.
> 0: system.remote_gdb: listening for remote gdb on port 7000
> **** REAL SIMULATION ****
> info: Entering event queue @ 0.  Starting simulation...
> panic: Illegal instruction 0x3002a073 at pc 0x0000000080000040:
> Memory Usage: 681348 KBytes
> Program aborted at tick 16032
> --- BEGIN LIBC BACKTRACE ---
> ./build/RISCV/gem5.opt(_Z15print_backtracev+0x2c)[0x5571f6f80eac]
> ./build/RISCV/gem5.opt(_Z12abortHandleri+0x4a)[0x5571f6f9287a]
> /lib/x86_64-linux-gnu/libpthread.so.0(+0x12980)[0x7fdcbbf3d980]
> /lib/x86_64-linux-gnu/libc.so.6(gsignal+0xc7)[0x7fdcba74bfb7]
> /lib/x86_64-linux-gnu/libc.so.6(abort+0x141)[0x7fdcba74d921]
> ./build/RISCV/gem5.opt(+0xe29aff)[0x5571f6a54aff]
> ./build/RISCV/gem5.opt(_ZN8RiscvISA16IllegalInstFault8invokeSEEP13ThreadContextRK14RefCountingPtrI10StaticInstE+0x2e4)[0x5571f6b3e664]
> ./build/RISCV/gem5.opt(_ZN8RiscvISA10RiscvFault6invokeEP13ThreadContextRK14RefCountingPtrI10StaticInstE+0x140)[0x5571f6b3f4a0]
> ./build/RISCV/gem5.opt(_ZN13BaseSimpleCPU9advancePCERKSt10shared_ptrI9FaultBaseE+0x145)[0x5571f70ba6e5]
> ./build/RISCV/gem5.opt(_ZN15AtomicSimpleCPU4tickEv+0x4d5)[0x5571f70ae3f5]
> ./build/RISCV/gem5.opt(_ZN10EventQueue10serviceOneEv+0xa5)[0x5571f6f89305]
> ./build/RISCV/gem5.opt(_Z9doSimLoopP10EventQueue+0x87)[0x5571f6fabdc7]
> ./build/RISCV/gem5.opt(_Z8simulatem+0xcba)[0x5571f6face1a]
> ./build/RISCV/gem5.opt(+0x128dbc1)[0x5571f6eb8bc1]
> ./build/RISCV/gem5.opt(+0xe4a554)[0x5571f6a75554]
> /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x64d7)[0x7fdcbc1eac47]
> /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x7d8)[0x7fdcbc329908]
> /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x5bf6)[0x7fdcbc1ea366]
> /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x7d8)[0x7fdcbc329908]
> /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x5bf6)[0x7fdcbc1ea366]
> /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x7d8)[0x7fdcbc329908]
> /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x5bf6)[0x7fdcbc1ea366]
> /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x7d8)[0x7fdcbc329908]
> /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCode+0x19)[0x7fdcbc1e45d9]
> /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x6ac0)[0x7fdcbc1eb230]
> /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x7d8)[0x7fdcbc329908]
> /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x5bf6)[0x7fdcbc1ea366]
> /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x7d8)[0x7fdcbc329908]
> /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCode+0x19)[0x7fdcbc1e45d9]
> /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyRun_StringFlags+0x76)[0x7fdcbc2946f6]
> ./build/RISCV/gem5.opt(_Z6m5MainiPPc+0x83)[0x5571f6f91373]
> ./build/RISCV/gem5.opt(main+0x38)[0x5571f5fa3be8]
> --- END LIBC BACKTRACE ---
> 중지됨 (core dumped)

Has anyone encountered this error? Please share a solution.

glzhou97, 2023-04-05:
It is because  at pc 0x0000000080000040 it contains high priority instruction that Gem5 se model cannot execute. You may need to directly modify the benchmark c file and remove the code that creates this csr instruction. E.g. remove setStats in dhrystone_main.c: line 114 and 168

UmerShahidengr, 2023-04-27:
I am facing the same issue. @glzhou97 your fix doesn't work. Does anyone else know how to resolve this error?

yfxiao16, 2024-06-12:
Hi @UmerShahidengr, did you solve this error?