riscv-software-src/riscv-tests #332: illegal instruction exception

ayaz91, 2021-03-19:
Hi everyone,

I am having some trouble while trying to run these tests in qemu and gem5. Basically, the execution is throwing an "illegal instruction" exception when "csrs mstatus, t0" instruction is being executed in any of the tests. Binary dump does not show any problem (the machine code seems all legal and inst addresses seem aligned as well).

I wonder if anyone has any pointers on what might be wrong here?
Thanks

ayaz91, 2021-03-22:
So, looking more into it, it seems like the exception is raised because we are trying to read mstatus in u-mode. Are these tests only supposed to be run on spike and do not support a linux abi? It will be great if someone can confirm this!

Thanks

BSEE201905, 2024-05-14:
[image]
  While i am running the coremark benchmark as workload on gem5  I am facing this issue if any one has solution please let me know.

aswaterman, 2024-05-14:
Yes, the generated test binaries are meant to run in M-mode.  Spike can run them, but they are not at all specific to Spike.

WaseemEng, 2024-05-15:
Is there any guide that how can i change the privilege mode form user to machine in gem5 simulator? If I can change the mode above issue can be solved?

rosekunthara, 2024-08-06:
> Is there any guide that how can i change the privilege mode form user to machine in gem5 simulator? If I can change the mode above issue can be solved?

Were you able to resolve this?

waseem-10xe, 2024-08-06:
Yes we have to run the gem5 simulator in full system mode to solve above problem. In user mode it would only emulates the linux calls.