riscv-software-src/riscv-tests #343: 'make[1]: *** [rv32ui-v-simple] Error 1' when trying to build

georgeyhere, 2021-07-02:
I followed the instructions to build from repo on WSL2 but keep getting the following error:
```
mkdir -p benchmarks
make -C benchmarks -f /mnt/c/users/george/riscv-tests/benchmarks/Makefile src_dir=/mnt/c/users/george/riscv-tests/benchmarks XLEN=64
make[1]: Entering directory '/mnt/c/users/george/riscv-tests/benchmarks'
make[1]: Nothing to be done for 'default'.
make[1]: Leaving directory '/mnt/c/users/george/riscv-tests/benchmarks'
mkdir -p isa
make -C isa -f /mnt/c/users/george/riscv-tests/isa/Makefile src_dir=/mnt/c/users/george/riscv-tests/isa XLEN=64
make[1]: Entering directory '/mnt/c/users/george/riscv-tests/isa'
riscv64-unknown-elf-gcc -march=rv32g -mabi=ilp32 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -DENTROPY=0xf7930f7 -std=gnu99 -O2 -I/mnt/c/users/george/riscv-tests/isa/../env/v -I/mnt/c/users/george/riscv-tests/isa/macros/scalar -T/mnt/c/users/george/riscv-tests/isa/../env/v/link.ld /mnt/c/users/george/riscv-tests/isa/../env/v/entry.S /mnt/c/users/george/riscv-tests/isa/../env/v/*.c rv32ui/simple.S -o rv32ui-v-simple
/mnt/c/users/george/riscv-tests/isa/../env/v/vm.c: In function 'handle_fault':
/mnt/c/users/george/riscv-tests/isa/../env/v/vm.c:175:3: error: both arguments to '__builtin___clear_cache' must be pointers
  175 |   __builtin___clear_cache(0,0);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/mnt/c/users/george/riscv-tests/isa/Makefile:80: recipe for target 'rv32ui-v-simple' failed
make[1]: *** [rv32ui-v-simple] Error 1
make[1]: Leaving directory '/mnt/c/users/george/riscv-tests/isa'
Makefile:28: recipe for target 'isa' failed
make: *** [isa] Error 2
```
riscv32-unknown-elf is installed to /opt/riscv32, and the RISCV environment variable is set to the same directory. Any help would be greatly appreciated!

aswaterman, 2021-07-02:
Looks like GCC is getting stricter.  If you replace `0,0` with `NULL, NULL`, does it work?  If so, can you make a pull request?

pdonahue-ventana, 2021-07-02:
I ran into the same problem a few months ago.  The gcc manual says:
> Built-in Function: void __builtin___clear_cache (void *begin, void *end)
> 
> This function is used to flush the processor’s instruction cache for the region of memory between begin inclusive and end exclusive.

So this call is clearing the Icache from 0 (inclusive) to 0 (exclusive) which is undefined behavior.  My solution was to just use an old compiler but, of course, that's not really a solution.

georgeyhere, 2021-07-02:
No luck:
```
mkdir -p benchmarks
make -C benchmarks -f /mnt/c/Users/George/riscv-tests/benchmarks/Makefile src_dir=/mnt/c/Users/George/riscv-tests/benchmarks XLEN=64
make[1]: Entering directory '/mnt/c/Users/George/riscv-tests/benchmarks'
make[1]: Nothing to be done for 'default'.
make[1]: Leaving directory '/mnt/c/Users/George/riscv-tests/benchmarks'
mkdir -p isa
make -C isa -f /mnt/c/Users/George/riscv-tests/isa/Makefile src_dir=/mnt/c/Users/George/riscv-tests/isa XLEN=64
make[1]: Entering directory '/mnt/c/Users/George/riscv-tests/isa'
riscv64-unknown-elf-gcc -march=rv32g -mabi=ilp32 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -DENTROPY=0xf7930f7 -std=gnu99 -O2 -I/mnt/c/Users/George/riscv-tests/isa/../env/v -I/mnt/c/Users/George/riscv-tests/isa/macros/scalar -T/mnt/c/Users/George/riscv-tests/isa/../env/v/link.ld /mnt/c/Users/George/riscv-tests/isa/../env/v/entry.S /mnt/c/Users/George/riscv-tests/isa/../env/v/*.c rv32ui/simple.S -o rv32ui-v-simple
/mnt/c/Users/George/riscv-tests/isa/../env/v/vm.c: In function 'handle_fault':
/mnt/c/Users/George/riscv-tests/isa/../env/v/vm.c:175:3: error: both arguments to '__builtin___clear_cache' must be pointers
  175 |   __builtin___clear_cache(NULL,NULL);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/mnt/c/Users/George/riscv-tests/isa/Makefile:80: recipe for target 'rv32ui-v-simple' failed
make[1]: *** [rv32ui-v-simple] Error 1
make[1]: Leaving directory '/mnt/c/Users/George/riscv-tests/isa'
Makefile:28: recipe for target 'isa' failed
make: *** [isa] Error 2
```

georgeyhere, 2021-07-07:
> I ran into the same problem a few months ago. The gcc manual says:
>
>
> So this call is clearing the Icache from 0 (inclusive) to 0 (exclusive) which is undefined behavior. My solution was to just use [...]

Can I ask what compiler version you used? I installed GCC 4.8.5 and the error keeps popping up.

georgeyhere, 2021-07-07:
Deleting the offending line from vm.c fixed the issue.

closed 2021-07-07 by georgeyhere