riscv-software-src/riscv-tests #350: PTE_N is not defined in napot.S

pdonahue-ventana, 2021-07-20:
I get this on the new test:
```
rv64si/napot.S: Assembler messages:
rv64si/napot.S:45: Error: illegal operands `li a1,0x001|0x010|0x002|0x004|0x008|0x040|0x080|PTE_N'
```

The test has:
```
isa/rv64si/napot.S:  li a1, PTE_V | PTE_U | PTE_R | PTE_W | PTE_X | PTE_A | PTE_D | PTE_N
```
All of those are defined in env/encoding.h except for PTE_N.  I think that a change to riscv-test-env is necessary.

@daniellustig

aswaterman, 2021-07-20:
Fixed.

closed 2021-07-20 by aswaterman

daniellustig, 2021-07-20:
@aswaterman what is the general procedure for updating and synchronizing encoding.h across the various repos?  PTE_N has been in riscv-opcodes since January, but it seems propagation of encoding.h updates to riscv-isa-sim and riscv-test-env (and elsewhere) is done manually?  I had a similar question at https://github.com/riscv/riscv-isa-sim/pull/750#issue-692148561...sorry if I did things out of order somehow.

aswaterman, 2021-07-20:
@daniellustig I did it semi-manually to pull in only necessary changes so that I didn't accidentally break anything.  Ideally, we would do it fully automatically, but that would require me being able to build and test the debug tests, which I can't do in my current environment.

commit 133856db (2021-07-20): Bump env

Fixes #350