riscv-software-src/riscv-tests #354: Issue to simulate interrupt.c with spike

JuanJRom, 2021-08-02:
I'm trying to simulate the [interrupt.c](https://github.com/riscv/riscv-tests/blob/master/debug/programs/interrupt.c) program. I have installed the following risc-v tools: **riscv64-unknown-elf-gcc**, **spike**, and **pk**.

The compilation is done without errors, but when simulating it generates the following error:

` jjrh@ubuntu-20:~/risc-v/Programs$ riscv64-unknown-elf-gcc -march=rv64g -o insterrupt insterrupt.c -static-libgcc -lm`
` jjrh@ubuntu-20:~/risc-v/Programs$ spike pk insterrupt`

` bbl loader`
`z  0000000000000000 ra 00000000000101e0 sp 0000003ffffffae0 gp 000000000001edc0
tp 0000000000000000 t0 0000000000000000 t1 000000000000000f t2 0000000000000000
s0 0000003ffffffb10 s1 0000000000000000 a0 0000000000000000 a1 0000000000000000
a2 0000000000000000 a3 0000000000000000 a4 0000000000000001 a5 0000000000000000
a6 000000000000001f a7 0000000000000000 s2 0000000000000000 s3 0000000000000000
s4 0000000000000000 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000
s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000
t3 0000000000000000 t4 0000000000000000 t5 0000000000000000 t6 0000000000000000
pc 000000000001016c va/inst 00000000304027f3 sr 8000000200006020`
`An illegal instruction was executed!`

In order to use interrupts, do I have to install another tool or why is this error being generated?

aswaterman, 2021-08-02:
Programs that run on top of `pk` run in U-mode, meaning you can't access M-mode registers.  You'd need to run bare metal (i.e. not on top of `pk`).

JuanJRom, 2021-08-03:
I already understand what the problem is.

Thanks a lot!

closed 2021-08-03 by JuanJRom