riscv-software-src/riscv-tests #356: rv64si/csr.S: does the test for mstatus.FS violate the RISC-V spec?

MikeDoingThings, 2021-08-16:
In test case 13, mstatus.FS is set to 0 before an fsw is tested. The behavior expected by the test program seems to be that the fsw instruction executes without an exception but does not perform the memory access, hence the data in memory is not changed. This seems to not be in line with the priviledged spec (p. 24): 

"When an extension’s status is set to Off, any instruction that attempts to read or write the corresponding state will cause an illegal instruction exception." 

Am I misinterpreting this line? Is a store instruction not a read or write access of the F-extension's state?

aswaterman, 2021-08-16:
Your interpretation of the ISA spec is correct, but the test is also correct, because it actually expects the `fsw` to trap.  See the trap handler at line 144, which anticipates this event.

MikeDoingThings, 2021-08-18:
You are right. My problem was connected to something else. This issue can be closed. 

Thank you very much for your help!

closed 2021-08-18 by MikeDoingThings