riscv-software-src/riscv-tests #357: Question about setting mstatus.mpp

Lemover, 2021-08-22:
In file "isa/rv64si/dirty.S" line 30, the program wants set the MMP to S-mode by a instruction CSRRS. However, it may fail to do that for two reasons:
1. In the macro **RVTEST_RV64M**, the program will set MPP to M-mode, which is b11.
2.  According to the riscv isa spec, CSRRS do the job: **The initial value in integer register rs1 is treated as a bit mask that specifies bit positions to be set in the CSR. Any bit that is high in rs1 will cause the corresponding bit to be set in the CSR, if that CSR bit is writable**. 
So the CSRRS instruction will do the OR operation. It will not change the MPP to b01, which is already b11.

There must be a bug in the instruction above or a misunderstanding of the riscv spec.

aswaterman, 2021-08-22:
I suggest stepping through execution of these tests on Spike to understand how they work before assuming they or the spec are buggy. In this case, the detail you missed is that MRET causes MPP to become 0 before the CSRRS instruction is executed.

Lemover, 2021-08-22:
Sorry for my straight words. I mean I may have a misunderstanding about the riscv spec (which i may not realize currently).

*Here is how i found the bug*:

I tries to run rv64si-p-dirty on spike in interactive mode, and got the result:
(a part of the logs, for some instructions don't output logs and no branch instrcutions executed)
command: **spike -d --pc=0x8000000 rv64si-p-dirty**
```
: pc 0
0x0000000000001000
: run 10
core   0: 0x0000000000001000 (0x00000297) auipc   t0, 0x0
core   0: 0x0000000000001004 (0x02028593) addi    a1, t0, 32
core   0: 0x0000000000001008 (0xf1402573) csrr    a0, mhartid
core   0: 0x000000000000100c (0x0182b283) ld      t0, 24(t0)
core   0: 0x0000000000001010 (0x00028067) jr      t0
core   0: 0x0000000080000000 (0x0040006f) j       pc + 0x4
core   0: 0x0000000080000004 (0x00000093) li      ra, 0
core   0: 0x0000000080000008 (0x00000113) li      sp, 0
core   0: 0x000000008000000c (0x00000193) li      gp, 0
core   0: 0x0000000080000010 (0x00000213) li      tp, 0
# keep run 1...
: run 1
core   0: 0x00000000800000bc (0x00200193) li      gp, 2
: run 1
core   0: 0x00000000800000c0 (0x00100393) li      t2, 1
: run 1
core   0: 0x00000000800000c4 (0x80002517) auipc   a0, 0x80002
: run 1
core   0: 0x00000000800000c8 (0xf4752223) sw      t2, -188(a0)
core   0: exception trap_store_access_fault, epc 0x00000000800000c8
core   0:           tval 0x0000000000002008
: run 1
core   0: exception trap_instruction_access_fault, epc 0x0000000000000000
core   0:           tval 0x0000000000000000
: run 1
core   0: exception trap_instruction_access_fault, epc 0x0000000000000000
core   0:           tval 0x0000000000000000
:
```
here is a part of the dump of rv64si-p-dirty:
```

Disassembly of section .text.init:

0000000080000000 <_start>:
    80000000: 0040006f            j 80000004 <reset_vector>

0000000080000004 <reset_vector>:
    80000004: 00000093            li  ra,0
    80000008: 00000113            li  sp,0
    8000000c: 00000193            li  gp,0
    80000010: 00000213            li  tp,0
    80000014: 00000293            li  t0,0
    80000018: 00000313            li  t1,0
    8000001c: 00000393            li  t2,0
    80000020: 00000413            li  s0,0
    80000024: 00000493            li  s1,0
    80000028: 00000513            li  a0,0
    8000002c: 00000593            li  a1,0
    80000030: 00000613            li  a2,0
    80000034: 00000693            li  a3,0
    80000038: 00000713            li  a4,0
    8000003c: 00000793            li  a5,0
    80000040: 00000813            li  a6,0
    80000044: 00000893            li  a7,0
    80000048: 00000913            li  s2,0
    8000004c: 00000993            li  s3,0
    80000050: 00000a13            li  s4,0
    80000054: 00000a93            li  s5,0
    80000058: 00000b13            li  s6,0
    8000005c: 00000b93            li  s7,0
    80000060: 00000c13            li  s8,0
    80000064: 00000c93            li  s9,0
    80000068: 00000d13            li  s10,0
    8000006c: 00000d93            li  s11,0
    80000070: 00000e13            li  t3,0
    80000074: 00000e93            li  t4,0
    80000078: 00000f13            li  t5,0
    8000007c: 00000f93            li  t6,0
    80000080: 00000193            li  gp,0
    80000084: 00002537            lui a0,0x2
    80000088: 8005051b            addiw a0,a0,-2048
    8000008c: 30052073            csrs  mstatus,a0
    80000090: fff0051b            addiw a0,zero,-1
    80000094: 03f51513            slli  a0,a0,0x3f
    80000098: 00002597            auipc a1,0x2
    8000009c: f805b583            ld  a1,-128(a1) # 80002018 <_GLOBAL_OFFSET_TABLE_+0x8>
    800000a0: 00c5d593            srli  a1,a1,0xc
    800000a4: 00a5e5b3            or  a1,a1,a0
    800000a8: 18059073            csrw  satp,a1
    800000ac: 12000073            sfence.vma
    800000b0: 000215b7            lui a1,0x21
    800000b4: 8005859b            addiw a1,a1,-2048
    800000b8: 3005a073            csrs  mstatus,a1
    800000bc: 00200193            li  gp,2
    800000c0: 00100393            li  t2,1
    800000c4: 80002517            auipc a0,0x80002
    800000c8: f4752223            sw  t2,-188(a0) # 2008 <_start-0x7fffdff8>
    800000cc: 00300193            li  gp,3
    800000d0: 000415b7            lui a1,0x41
    800000d4: 8005859b            addiw a1,a1,-2048
    800000d8: 3005a073            csrs  mstatus,a1
    800000dc: 80002297            auipc t0,0x80002
    800000e0: f2c2a283            lw  t0,-212(t0) # 2008 <_start-0x7fffdff8>
# ...
```

In spike interactive interface, it seems "difficult" to see csr.mstatus' value. So I don't know how the mstatus changes. And my understanding about the program is:
```
...
    8000008c: 30052073            csrs  mstatus,a0 // sets the mstatus.mpp to b11. In RVTEST_RV64M
...
    800000b8: 3005a073            csrs  mstatus,a1 // wants to set mstatus.mpp to b01 (S-mode). but failed
...
    800000c8: f4752223            sw  t2,-188(a0) # 2008 <_start-0x7fffdff8> // wants to store to vaddr 0x2008, but failed, because vm is not enabled for mstatus.mpp is still b11. As a result, it fails to translate and raises a access fault, not a page fault.
...
```
The MRET has not yet executed, so it faile to reset MPP to 0.

I realized that the mepc is not set correctly, so i turn to look at the macros. And i find that the macro in my branch is different from the master. The code has been changed long ago and some init code was changed.

Thank u for the instruction.

closed 2021-08-22 by Lemover