riscv-software-src/riscv-tests #368: Removing CSR dependency from rv32ui and rv64ui tests

nindanaoto, 2022-01-22:
Dear maintainers,

As you may know, Control State Register (CSR) is removed from base instruction in RV32I and RV64I version 2.1. So, we should remove the use of CSR related instructions in rv32ui and rv64ui tests. If not, at least, README should mention that rv32ui and rv64ui tests are tests for base Integer instructions + Zicsr exetension.

tudortimi, 2022-11-10:
I also see that the tests make use of the `mret` instruction, which is defined in the privileged spec.

Tastaturtaste, 2025-05-01:
I also ran into this issue and lost a few hours of my life on this. Could this please _at least_ be mentioned in the readme?

codygunton, 2025-05-16:
@timsifive @aswaterman I'm considering resolving this issue, but I'm afraid it requires a lot of context on a broad range of RISC-V issues that I don't have. I'd greatly appreciate it if you could outline a strategy that you (or any other contributors reading this!) would be happy to see implemented and point out any potential gotchas.

codygunton, 2025-05-16:
Thanks, yes, so for context, I am interested in running these to test purely virtual implementations of rv32im (cf [this discussion](https://github.com/risc0/risc0/discussions/3160)) that are used to make cryptographic proofs of RISC-V execution. The types of programs that are of interest in that context are not programs running on a conventional OS, so such zk cryptography libraries don't implement extensions like Zicsr. As you can see in links ([e.g.](https://github.com/risc0/zirgen/blob/main/zirgen/circuit/rv32im/v1/test/riscv_test.h)) there are workarounds in use, but I'm wondering if it makes sense to contribute to the RISC-V ecosystem more broadly and resolve this issue. Rather than dive in I thought I'd seek guidance on a strategy for updating the testing framework, or perhaps some input that it's not desirable to do so. Thanks again!

aswaterman, 2025-05-16:
The sorts of workarounds you mention are exactly the approach I'd recommend--replace those macro definitions with something appropriate for your platform.

I'd have to see the results to form my own opinion on whether or not it would be useful for the broader community.  Just as the existing test environments encode some details of the platform modeled by Spike, it might be the case that your test environment encodes details of your platform that limit its applicability for other users.