riscv-software-src/riscv-tests #375: rv64mi/access.S - JALR assumes non-precise misaligned exception behavior.

john-terrell, 2022-03-06:
The code inside RV64/mi/access.S contains this comment:

      # jalr to an illegal address should commit (hence should write rd).
      # after the pc is set to rs1, an access exception should be raised.

However, in discussions, it appears the JALR misaligned exception should be precise (and *not* write RD):

    "Misaligned address trap should be taken precisely, implying rd would not be updated. This is same as for any architectural state update for a precise trap, so trying to clarify here would cause confusion everywhere else (or require every instruction description repeated what precise trap means)." - kasanovic (https://github.com/riscv/riscv-isa-manual/issues/269)

Is my understanding on this correct?   Is the access.S test for RV64i in conflict with the desired (precise) behavior of misaligned instruction exceptions?

Thanks.

aswaterman, 2022-03-08:
Your understanding of misaligned traps is correct, but they are a special case, because they are easily detected when executing the JALR instruction itself.  Other address exceptions require attempting to fetch the target instruction, and so they are reported on the target instruction.  As the spec says: "If an instruction access-fault or instruction page-fault exception occurs on the target of a jump or taken branch, the exception is reported on the target instruction, not on the jump or branch instruction."

The exception is still precise, but it is precise with respect to the following instruction.

closed 2022-03-08 by aswaterman

john-terrell, 2022-03-08:
Thanks for the clarification!