riscv-software-src/riscv-tests #377: quick smoke debug test against spike failed

whutddk, 2022-03-29:
I try to run a quick smoke debug test against spike, by running `make`, but it seems that the openocd cannot connect to the target:

the log is as follows:

```
Test: CheckMisa
Target: spike64
---------------------------------[ Compile ]----------------------------------
+ riscv64-unknown-elf-gcc -g programs/checksum.c programs/tiny-malloc.c programs/infinite_loop.S -DDEFINE_MALLOC -DDEFINE_FREE programs/entry.S programs/init.c -DNHARTS=1 -I ../env -T targets/RISC-V/spike64.lds -nostartfiles -mcmodel=medany -DXLEN=64 -o /tmp/spike64_checksum-8000000000141125_yippmk5c -march=rv64imafc -mabi=lp64
-------------------------[ /tmp/openocdl1eih6so.log ]-------------------------
+ REMOTE_BITBANG_HOST=localhost REMOTE_BITBANG_PORT=61104 WORK_AREA=0x1212340000 USE_FREERTOS=0 /RISCV//bin/openocd --command 'gdb_port 0' --command 'tcl_port disabled' --command 'telnet_port disabled' -f targets/RISC-V/spike-1.cfg
Open On-Chip Debugger 0.10.0+dev-00198-g35eed36ff (2022-03-29-16:06)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 10000 kHz
Warn : Adapter driver 'remote_bitbang' did not declare which transports it allows; assuming legacy JTAG-only
Info : only one transport option; autoselect 'jtag'
Info : Initializing remote_bitbang driver
Info : Connecting to localhost:61104
Info : remote_bitbang driver initialized
Info : This adapter doesn't support configurable speed
Info : JTAG tap: riscv.cpu tap/device found: 0xdeadbeef (mfg: 0x777 (<unknown>), part: 0xeadb, ver: 0xd)
Warn : JTAG tap: riscv.cpu       UNEXPECTED: 0xdeadbeef (mfg: 0x777 (<unknown>), part: 0xeadb, ver: 0xd)
Error: JTAG tap: riscv.cpu  expected 1 of 1: 0x10e31913 (mfg: 0x489 (SiFive, Inc.), part: 0x0e31, ver: 0x1)
Error: Trying to use configured scan chain anyway...
Warn : Bypassing JTAG setup events due to errors
Error: Debugger is not authenticated to target Debug Module. (dmstatus=0x4c0c02). Use `riscv authdata_read` and `riscv authdata_write` commands to authenticate.
Info : Listening on port 61106 for gdb connections
0x327b23c6
Error: Target not examined yet

-------------------------[ /tmp/spike-ndaclf_1.log ]--------------------------
+ /RISCV//bin/spike -p1 --isa RV64IMAFC --dm-auth --dm-progsize 0 --dm-sba 64 --dm-abstract-rti 30 -m0x1212340000:0x10000000 --rbb-port 0 /tmp/spike64_checksum-8000000000141125_yippmk5c
Listening for remote bitbang connection on port 61104.
warning: tohost and fromhost symbols not in ELF; can't communicate with target
Received nothing. Quitting.

--------------------------------[ Traceback ]---------------------------------
Traceback (most recent call last):
  File "/mnt/c/work/riscv-tools/riscv-tests/debug/testlib.py", line 1114, in run
    self.classSetup()
  File "/mnt/c/work/riscv-tools/riscv-tests/debug/testlib.py", line 1166, in classSetup
    BaseTest.classSetup(self)
  File "/mnt/c/work/riscv-tools/riscv-tests/debug/testlib.py", line 1081, in classSetup
    self.server = self.target.server(self)
  File "/mnt/c/work/riscv-tools/riscv-tests/debug/targets.py", line 166, in server
    return testlib.Openocd(server_cmd=self.server_cmd,
  File "/mnt/c/work/riscv-tools/riscv-tests/debug/testlib.py", line 349, in __init__
    self.process = self.start(cmd, logfile, extra_env)
  File "/mnt/c/work/riscv-tools/riscv-tests/debug/testlib.py", line 368, in start
    raise Exception("OpenOCD exited early.")
Exception: OpenOCD exited early.
-------------------------[ /tmp/spike-ndaclf_1.log ]--------------------------
+ /RISCV//bin/spike -p1 --isa RV64IMAFC --dm-auth --dm-progsize 0 --dm-sba 64 --dm-abstract-rti 30 -m0x1212340000:0x10000000 --rbb-port 0 /tmp/spike64_checksum-8000000000141125_yippmk5c
Listening for remote bitbang connection on port 61104.
warning: tohost and fromhost symbols not in ELF; can't communicate with target
Received nothing. Quitting.

-------------------------------[ End of logs ]--------------------------------
Result: exception
Logfile: logs/20220329-185435-spike64-CheckMisa.log
Reproduce: ./gdbserver.py ./targets/RISC-V/spike64.py CheckMisa
Time elapsed: 0.99s
```

I am using Ubuntu20 in WSL2. Does it matter?

timsifive, 2022-03-29:
> Open On-Chip Debugger 0.10.0+dev-00198-g35eed36ff (2022-03-29-16:06)

That OpenOCD seems kind of old (0.10, even though it was built recently). The tests are intended to run with OpenOCD from https://github.com/riscv/riscv-openocd.

whutddk, 2022-03-30:
I downloaded a new prebuilt openocd from SiFive, and got fewer errors and most of the tests passed. 
Thank you!!!

closed 2022-03-30 by whutddk

hyperion009, 2024-01-02:
@timsifive Hi Tim， I have the same issue with openocd-0.12.0+dev-03368-gf119c1d48-dirty (2023-11-09-13:39) which is downloaded from the master branch of https://github.com/riscv/riscv-openocd.
[image]
@whutddk Hi whutddk, do you download from https://github.com/sifive/freedom-tools?
thanks

whutddk, 2024-01-02:
https://www.sifive.com/software
Here.