riscv-software-src/riscv-tests #389: ma_addr test fails if mtval is hardwired to 0

Wren6991, 2022-05-28:
This trap handler in ma_addr.S checks that hardware sets `mtval` to the load/store address on a load/store alignment exception:

https://github.com/riscv-software-src/riscv-tests/blob/fd6e7ed1707b3449f9673fa6b3bc1e422af4669b/isa/rv64mi/ma_addr.S#L97-L108

However hardware is not obligated to do anything more than implement the register as hardwired to 0, as per the latest priv spec:

[image]

With the following patch, this test passes on my implementation (which hardwires `mtval` to 0), and is still a useful test, since the actual trap is still exercised:

```patch
diff --git a/isa/rv64mi/ma_addr.S b/isa/rv64mi/ma_addr.S
index 721ac6a..f02a1af 100644
--- a/isa/rv64mi/ma_addr.S
+++ b/isa/rv64mi/ma_addr.S
@@ -99,10 +99,12 @@ mtvec_handler:
   bne t0, s1, fail
 
   csrr t0, mbadaddr
+  beqz t0, 1f
   bne t0, t1, fail
 
   lb t0, (t0)
   beqz t0, fail
+1:
 
   csrw mepc, t2
   mret
```

This probably isn't good enough for upstream, though, because it weakens the test for implementations which attempt to set `mtval` informatively.

I'm not sure what the best solution is. Maybe it's enough to assert that either all loads/stores set `mtval` informatively, or all loads/stores set `mtval` to 0? I'm happy to create any appropriate PRs.

aswaterman, 2022-05-28:
Not executing the test on implementations that hardwire mtval to 0 is also an option. But I’d be OK with your proposed patch, too.

Wren6991, 2022-05-28:
Okay, I raised a PR for the simple patch. I'm not looking for trouble.

aswaterman, 2022-05-29:
No trouble! Thanks for the suggestion.

commit 3dc79832 (2022-05-29): Permit mtval to be zero in misaligned address test, fixes #389 (#390)
closed 2022-05-29 by aswaterman