riscv-software-src/riscv-tests #394: exceptions in p and v environments

timsifive, 2022-06-07:
I'm writing tests for misaligned accesses, which requires the exception to be handled in the test if the hardware doesn't support misaligned accesses. This works great in the p test environment (copying the pattern from breakpoint.S), but not at all in the v environment. These environments look completely different. In p, it calls mtvec_handler (if it's defined) from trap_vector, in p/riscv_test.h. In v, it jumps to wtf from trap_vector, defined in v/entry.S.

Why do these two environments handle this so differently, to the point where they even put the functionality in different files? Can I make the v environment more like the p environment to deal with this?

timsifive, 2022-06-07:
I could `#define EXCEPTION_FREE 1` in v/riscv_test.h

aswaterman, 2022-06-07:
Please don't.  Tests meant to execute in that VM are exception-free.  Tests that might cause exceptions just shouldn't run in that VM.  The test should be moved to a directory like `rv32mi` etc.; those VMs permit exceptions.

closed 2022-06-07 by timsifive