riscv-software-src/riscv-tests #453: PMP example doesn't seem to work

pcotret, 2023-02-10:
In the PMP example, the `sptbr` register is used to initialize the page table:
https://github.com/riscv-software-src/riscv-tests/blob/f163ddfc9b0bf78fb8180012ffcfc93427f780f0/benchmarks/pmp/pmp.c#L46-L47

In RISC-V specs (https://github.com/riscv/riscv-isa-manual/releases/tag/draft-20230131-c0b298a release) :
Preface to 1.10:
> The machine-mode base-and-bounds translation and protection schemes have been removed from the specification as part of moving the virtual memory configuration to `sptbr` (**now `satp`**). Some of the motivation for the base and bound schemes are now covered by the PMP registers, but space remains available in mstatus to add these back at a later date if deemed useful.

In the `env` submodule, the new name is clearly there :
https://github.com/riscv/riscv-test-env/blob/982f93f5c55f6e7931c01afb082f5ca42cffddab/encoding.h#L1644

aswaterman, 2023-02-10:
Some assemblers include the old name for backward compatibility, which is why it hasn’t been an issue. Of course, we’d accept a PR that updates the choice of name.

aswaterman, 2023-02-13:
Resolved with #455

closed 2023-02-13 by aswaterman