riscv-software-src/riscv-tests #482: Many debug test programs fail to compile due to lack of `zicsr` extension in `-march=...`

TommyMurphyTM1234, 2023-06-29:
Many test programs use CSR read/write instructions but are compiled using a `-march=...` option that does not include the `zicsr` extension. This was not a problem in the past (pre GCC 12, I think) when the `zicsr` and `zifencei` extensions were part of the base integer ISA. However that is no longer the case and programs using CSR read/write instructions must be compiled with a `-march=...` that includes the `zicsr` extension (e.g. `-zicsr=rv32imac_zicsr` or perhaps `-march=rv32gc` which still seems to - actually or effectively - get expanded to `-march=rv32imafdc_zicsr_zifencei`) otherwise compilation will fail with errors like this:
```
programs/entry.S:30: Error: unrecognized opcode `csrw mtvec,t0', extension `zicsr' required
```
See also here:

* https://github.com/riscv/riscv-openocd/issues/869#issuecomment-1613014637

timsifive, 2023-06-29:
Good find. Do you have time to look at this? I imagine it's just a few lines changed near https://github.com/riscv-software-src/riscv-tests/blob/master/debug/targets.py#L198

TommyMurphyTM1234, 2023-06-29:
At a glance it seems to me that changing these two lines:

* https://github.com/riscv-software-src/riscv-tests/blob/784c89ee1c4e7124a0973fda744522ebe532174b/debug/targets.py#L199
* https://github.com/riscv-software-src/riscv-tests/blob/784c89ee1c4e7124a0973fda744522ebe532174b/debug/targets.py#L209

to

* `          args.append("-march=rv32e_zicsr")`
* `          args.append(f"-march={march}_zicsr")`

might be sufficient?

timsifive, 2023-06-29:
That sounds totally believable. There's not much complexity going on with the compiling. I'm not aware of anyone running these tests against anything besides spike, and now also HiFive1.

TommyMurphyTM1234, 2023-06-29:
Ok, thanks @timsifive.
I'll submit a PR for this minor change tomorrow.

TommyMurphyTM1234, 2023-06-30:
As mentioned in the PR, this change means that the tests will most likely fail to compile with a pre GCC 12 RISC-V toolchain that predates the separation of the `zicsr` (and `zifencei`) extension(s) out from the base integer ISA because it will not understand `-march=rvXX..._zicsr`. But I didn't think that it was worth complicating the script further in order to try to detect and deal with such older toolchains.

TommyMurphyTM1234, 2023-07-05:
Fixed by https://github.com/riscv-software-src/riscv-tests/pull/484

closed 2023-07-05 by TommyMurphyTM1234