riscv-software-src/riscv-tests #504: Is it possible to modify the timer address to be configurable?

lz-bro, 2023-09-26:
The address of the mtime and mtimecmp  is different in different targets. It is better to configure it through user configuration.
https://github.com/riscv-software-src/riscv-tests/blob/d4eaa5bd6674b51d3b9b24913713c4638e99cdd9/debug/programs/init.h#L4-L5
https://github.com/riscv-software-src/riscv-tests/blob/d4eaa5bd6674b51d3b9b24913713c4638e99cdd9/debug/programs/run_halt_timing.S#L14
https://github.com/riscv-software-src/riscv-tests/blob/d4eaa5bd6674b51d3b9b24913713c4638e99cdd9/debug/gdbserver.py#L1049-L1050
https://github.com/riscv-software-src/riscv-tests/blob/d4eaa5bd6674b51d3b9b24913713c4638e99cdd9/debug/gdbserver.py#L915
In addition, SimpleNoExistTest uses a fixed register csr2288, which can be modified to be configurable
https://github.com/riscv-software-src/riscv-tests/blob/d4eaa5bd6674b51d3b9b24913713c4638e99cdd9/debug/gdbserver.py#L222-L233

timsifive, 2023-09-26:
I agree. The debug tests would be better if the mtime address and the CSR that doesn't exist would be configurable. I don't have time to work on this right now.

lz-bro, 2023-10-10:
Hi, @timsifive.  If I want to achieve configurable, what should I do better, can you give some suggestions?

timsifive, 2023-10-10:
In targets.py, modify the Hart class to add properties for mtime/mtimecmp address. Probably easiest to default to the spike values.

In Target.do_compile(), add -DMTIME_ADDR=... flags to the compiler.

Modify run_halt_timing.S to use that constant instead of the hard-coded value.

In gdbserver.py, get the address from the hart object.

Once you've done that, the CSR change would be similar but probably a little simpler since we don't have any compiled tests that reference them.

lz-bro, 2023-10-11:
Thank you very much for your suggestions.

timsifive, 2023-10-11:
Actually, mtime/mtimecmp might fit better in the Target than the Hart class. Generally memory-mapped registers all have the same address everywhere, and don't differ per hart.

lz-bro, 2023-10-12:
When I did FPGA debug testing, I found that our CLINT address was not 0x02000000.

lz-bro, 2023-10-13:
I may have misunderstood you at first. You meant that the harts in the target have the same memory-mapped registers, so it is better to define mtime/mtimecmp in the Target than the Hart class.

timsifive, 2023-10-13:
> I may have misunderstood you at first. You meant that the harts in the target have the same memory-mapped registers, so it is better to define mtime/mtimecmp in the Target than the Hart class.

Yes, that is what I meant. Sorry if it wasn't clear.

closed 2023-11-16 by lz-bro