riscv-software-src/riscv-tests #516: Does the IcountTest need to be improved?

lz-bro, 2023-10-17:
https://github.com/riscv-software-src/riscv-tests/blob/c36c814ae49610bc407a5c4e144866fa50e4510e/debug/gdbserver.py#L2059-L2083

In this test, The count are set to 1 and 2 respectively. But, it is mentioned in RISC-V External Debug Support Version 0.13 that the count may only support 1.
[image]
If the hardware only supports count 1, but if count was set to 2, the hardware will process count to 1, causing set_trigger fail due to inconsistent writing and reading back of tdata1.

timsifive, 2023-10-17:
Most of these tests are written to run well against spike, and unfortunately not against more general hardware. I think it's great that you're going through and noting some of these issues.

What should happen on hardware that supports count=1 but not count=2 is that `riscv icount set m 2` returns an error.  OpenOCD will read back tdata1 after writing it, and should return an error because the hardware doesn't let count=2 be written.

From the test framework, this means that we need a hart property: max_icount or something. Then in the test we can confirm that either icount works, or we get an error when setting it.

lz-bro, 2023-10-18:
Thank you very much for your reply and suggestions.

closed 2023-11-16 by lz-bro