riscv-software-src/riscv-tests #517: Is it necessary to add a SysbusTest class to test that using memory access by system bus?

lz-bro, 2023-10-17:


timsifive, 2023-10-17:
For context, the way I get OpenOCD coverage is because `make all` in some of its spike configurations runs a spike that doesn't support the program buffer at all.

If you want system bus coverage for your hardware, I would create a new .py file for that hardware, which uses an OpenOCD config script that uses `riscv set_mem_access sysbus` which should result in OpenOCD only using SBA memory accesses.

lz-bro, 2023-10-18:
Do you mean to use an OpenOCD config script that uses `riscv set_mem_access sysbus` to act on all test cases? This will lead to some test cases errors. For example, the address of mtime cannot be accessed using sysbus.
My idea is that is it possible to add a class `SysbusTest` in gdbserver.py to test system bus access individually?
int class SysbusTest, I can use `monitor riscv set_mem_access sysbus` to result in OpenOCD only using SBA memory accesses, and use `monitor default_mem_access` which was defined in an OpenOCD config script to reset.
However, I don’t have a good idea to design the class SysbusTest.

timsifive, 2023-10-19:
Hm... Such a target could set the mtime address to None, indicating that tests that need mtime should report not_applicable.

It's not a terrible idea to add a SysbusTest (and probably an AbstractMemTest also). They would probably do exactly what the current MemTest* tests do, but setting `set_mem_access` first. It would provide a decent smoke test for those features, although if it were my hardware I'd still also want to test more comprehensively with a separate target as I described earlier.

lz-bro, 2023-10-20:
What you said is very reasonable. Thank you for your explanation
.

closed 2023-11-16 by lz-bro