riscv-software-src/riscv-tests #523: how to use debug tools during simulation stage？

zhajio1988, 2023-12-22:
Such as i want to use debug feature of risc-tests during Synopsys VCS or Cadence  XRUN simulation?

timsifive, 2023-12-26:
It's possible to do this, but I should warn you that these tests take a very long time to run in simulation. E.g. CheckMisa, which is one of the smaller tests, scans over 13000 JTAG cycles.

Having said that, what I've seen done with a rocket core is that you implement https://github.com/fjullien/jtag_vpi in your simulation, and then write an OpenOCD config file that connects to that. You'll have to build an OpenOCD with `--enable-jtag_vpi`. Then make a python config file following the examples in the targets/ directory.

zhajio1988, 2024-03-15:
> It's possible to do this, but I should warn you that these tests take a very long time to run in simulation. E.g. CheckMisa, which is one of the smaller tests, scans over 13000 JTAG cycles.
>
> Having [...]

Thanks for your replay.

zhajio1988, 2024-03-15:
Hi，
After a day of research I have found the problem, I modified my own target by referring to targets/SiFive/Freedom/U500Sim.py, but there is a problem in this script 
```
def target(self).
      return testlib.VcsSim(sim_cmd=self.sim_cmd, debug=False)
```
It should read
```
def create(self): return testlib.
      return testlib.VcsSim(sim_cmd=self.sim_cmd, debug=False)
```
Otherwise the VCS simulation won't start

Translated with www.DeepL.com/Translator (free version)

zhajio1988, 2024-03-18:
@timsifive 
As previous comment, i have run some testcase successfully. But i also found a little issue. If I set the jtag's clk to 200 MHZ, CheckMisa is wrong, if I set it to 100 MHZ, CheckMisa passes! Could you help confirm the max clock of the jtag during debug testing?

zhajio1988, 2024-03-18:
In addition, these cases, each case run how long, I currently run to DebugFunctionCall, this case has been running for more than ten hours!