riscv-software-src/riscv-tests #524: How to configure 'expose_csrs' in a jtag-chain structure with multiple taps

lz-bro, 2023-12-27:
my config is as follows

```
adapter driver jlink
transport select jtag
adapter speed 2000

bindto 0.0.0.0
gdb_port  1024

set _CHIPNAME riscv

jtag newtap pre unknown  -irlen 1 -expected-id 0x00000000 -disable
jtag configure pre.unknown -event tap-enable ""

jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id 0x10000E21 -disable
jtag configure $_CHIPNAME.cpu -event tap-enable ""

jtag newtap post unknown -irlen 9 -expected-id 0x08501C0D

set _TARGETNAME_0 $_CHIPNAME.cpu0
set _TARGETNAME_1 $_CHIPNAME.cpu1

target create $_TARGETNAME_0 riscv -chain-position $_CHIPNAME.cpu -rtos hwthread
target create $_TARGETNAME_1 riscv -chain-position $_CHIPNAME.cpu -coreid 1

jtag configure post.unknown -event setup {
    global _CHIPNAME

    irscan post.unknown 0x98
    drscan post.unknown 16 0xa

    jtag tapenable pre.unknown
    jtag tapenable $_CHIPNAME.cpu
}

foreach t [target names] {
    targets $t
    riscv set_mem_access progbuf
    riscv expose_csrs 776,1984
}

init

set challenge [riscv authdata_read]
riscv authdata_write [expr {$challenge + 1}]

foreach t [target names] {
    targets $t
    halt
    arm semihosting enable
}
```
log show that TAP riscv.cpu is disabled, can't be the current target
```
Debug: 97 3 command.c:155 script_debug(): command - target names
Debug: 98 3 command.c:155 script_debug(): command - targets riscv.cpu0
Debug: 99 3 command.c:544 run_command(): Command 'targets' failed with error code -4
User : 100 3 command.c:608 command_run_line(): Target: TAP riscv.cpu is disabled, can't be the current target

    TargetName         Type       Endian TapName            State
--  ------------------ ---------- ------ ------------------ ------------
 0  riscv.cpu0         riscv      little riscv.cpu          tap-disabled
 1* riscv.cpu1         riscv      little riscv.cpu          tap-disabled
Debug: 101 3 riscv.c:492 riscv_deinit_target(): [riscv.cpu0] riscv_deinit_target()
Debug: 102 3 target.c:2203 target_free_all_working_areas_restore(): freeing all working areas
Debug: 103 3 riscv.c:492 riscv_deinit_target(): [riscv.cpu1] riscv_deinit_target()
Debug: 104 4 target.c:2203 target_free_all_working_areas_restore(): freeing all working areas
```
so I changed the config to

```
adapter driver jlink
transport select jtag
adapter speed 2000

bindto 0.0.0.0
gdb_port  1024

set _CHIPNAME riscv

jtag newtap pre unknown  -irlen 1 -expected-id 0x00000000 -disable
jtag configure pre.unknown -event tap-enable ""

jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id 0x10000E21 -disable
jtag configure $_CHIPNAME.cpu -event tap-enable ""

jtag newtap post unknown -irlen 9 -expected-id 0x08501C0D

set _TARGETNAME_0 $_CHIPNAME.cpu0
set _TARGETNAME_1 $_CHIPNAME.cpu1

target create $_TARGETNAME_0 riscv -chain-position $_CHIPNAME.cpu -rtos hwthread
target create $_TARGETNAME_1 riscv -chain-position $_CHIPNAME.cpu -coreid 1

jtag configure post.unknown -event setup {
    global _CHIPNAME

    irscan post.unknown 0x98
    drscan post.unknown 16 0xa

    jtag tapenable pre.unknown
    jtag tapenable $_CHIPNAME.cpu

    foreach t [target names] {
        targets $t
        riscv set_mem_access progbuf
        riscv expose_csrs 776,1984
    }

    init

    set challenge [riscv authdata_read]
    riscv authdata_write [expr {$challenge + 1}]

    foreach t [target names] {
        targets $t
        halt
        arm semihosting enable
    }
}
```
log show that the 'riscv expose_csrs' command must be used before 'init'.
```
Info : Hardware thread awareness created
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : J-Link V9 compiled May  7 2021 16:26:12
Info : Hardware version: 9.40
Info : VTarget = 3.320 V
Info : clock speed 2000 kHz
Info : JTAG tap: post.unknown tap/device found: 0x08501c0d (mfg: 0x606 (Shenzhen Chixingzhe Technology Co Ltd), part: 0x8501, ver: 0x0)
Info : JTAG tap: pre.unknown enabled
Info : JTAG tap: riscv.cpu enabled
Error: The 'riscv expose_csrs' command must be used before 'init'.
spacemit_x60_prijtag_1x2.cfg:38: Error: 
in procedure 'jtag_init' called at file "../src/jtag/core.c", line 1684
at file "spacemit_x60_prijtag_1x2.cfg", line 38
Info : [riscv.cpu0] datacount=2 progbufsize=2
Info : [riscv.cpu0] Vector support with vlenb=32
Info : [riscv.cpu0] Examined RISC-V core; found 4 harts
Info : [riscv.cpu0]  XLEN=64, misa=0x8000000000b4112f
[riscv.cpu0] Target successfully examined.
Info : [riscv.cpu1] datacount=2 progbufsize=2
Info : [riscv.cpu1] Vector support with vlenb=32
Info : [riscv.cpu1] Examined RISC-V core; found 4 harts
Info : [riscv.cpu1]  XLEN=64, misa=0x8000000000b4112f
[riscv.cpu1] Target successfully examined.
Info : starting gdb server for riscv.cpu0 on 1024
Info : Listening on port 1024 for gdb connections
Info : starting gdb server for riscv.cpu1 on 1025
Info : Listening on port 1025 for gdb connections
Info : [riscv.cpu0] became available (halted)
Info : [riscv.cpu1] became available (halted)
```

I hope someone can give me some help. Thanks!

en-sc, 2023-12-27:
If I'm correct, replacing
```
foreach t [target names] {
    targets $t
    riscv set_mem_access progbuf
    riscv expose_csrs 776,1984
}
```
with:
```
foreach t [target names] {
    $t riscv set_mem_access progbuf
    $t riscv expose_csrs 776,1984
}
```
in the first config (with `expose_csrs` before `init`) should work.

lz-bro, 2023-12-27:
Solved, thank you!!!

closed 2023-12-27 by lz-bro