riscv-software-src/riscv-tests #546: `--with-xlen=32` builds are failing

bluewww, 2024-03-08:
In `riscv-tests/benchmarks/Makefile`, the ABI is [hardcoded](https://github.com/riscv-software-src/riscv-tests/blob/master/benchmarks/Makefile#L46) to `lp64d`.

This causes  `XLEN=32` builds to fail like this

```bash
mkdir -p benchmarks
make -C benchmarks -f /tmp/riscv-tests/benchmarks/Makefile src_dir=/tmp/riscv-tests/benchmarks XLEN=32 
make[1]: Entering directory '/tmp/riscv-tests/benchmarks'
riscv32-unknown-elf-gcc -I/tmp/riscv-tests/benchmarks/../env -I/tmp/riscv-tests/benchmarks/common -I/tmp/riscv-tests/benchmarks/median -I/tmp/riscv-tests/benchmarks/qsort -I/tmp/riscv-tests/benchmarks/rsort -I/tmp/riscv-tests/benchmarks/towers -I/tmp/riscv-tests/benchmarks/vvadd -I/tmp/riscv-tests/benchmarks/memcpy -I/tmp/riscv-tests/benchmarks/multiply -I/tmp/riscv-tests/benchmarks/mm -I/tmp/riscv-tests/benchmarks/dhrystone -I/tmp/riscv-tests/benchmarks/spmv -I/tmp/riscv-tests/benchmarks/mt-vvadd -I/tmp/riscv-tests/benchmarks/mt-matmul -I/tmp/riscv-tests/benchmarks/mt-memcpy -I/tmp/riscv-tests/benchmarks/pmp -I/tmp/riscv-tests/benchmarks/vec-memcpy -I/tmp/riscv-tests/benchmarks/vec-daxpy -I/tmp/riscv-tests/benchmarks/vec-sgemm -I/tmp/riscv-tests/benchmarks/vec-strcmp -DPREALLOCATE=1 -mcmodel=medany -static -std=gnu99 -O2 -ffast-math -fno-common -fno-builtin-printf -fno-tree-loop-distribute-patterns -march=rv32gcv -mabi=lp64d -o median.riscv /tmp/riscv-tests/benchmarks/median/median_main.c /tmp/riscv-tests/benchmarks/median/median.c  /tmp/riscv-tests/benchmarks/common/syscalls.c /tmp/riscv-tests/benchmarks/common/crt.S -static -nostdlib -nostartfiles -lm -lgcc -T /tmp/riscv-tests/benchmarks/common/test.ld
cc1: error: ABI requires '-march=rv64'
cc1: error: ABI requires '-march=rv64'
cc1: error: ABI requires '-march=rv64'
cc1: error: ABI requires '-march=rv64'
make[1]: *** [/tmp/riscv-tests/benchmarks/Makefile:60: median.riscv] Error 1
make[1]: Leaving directory '/tmp/riscv-tests/benchmarks'
make: *** [Makefile:25: benchmarks] Error 2
```

TommyMurphyTM1234, 2024-06-27:
Can be closed now that this PR addresses the issue:

- https://github.com/riscv-software-src/riscv-tests/pull/566

(Unfortunately I don't have privileges to close the issue as resolved).

aswaterman, 2024-06-27:
Resolved by #566

closed 2024-06-27 by aswaterman