riscv-software-src/riscv-tests #548: `__riscv_flen` not defined, causing `rv64mi-p-csr` to fail

emturner, 2024-04-10:
Currently `rv64mi-p-csr` fails on our interpreter, due to line: https://github.com/riscv-software-src/riscv-tests/blob/6b1d7372d951ed75811e0a09c0fe9e065c141c2d/isa/rv64si/csr.S#L80

Could I please ask how we can compile with F enabled? all the other `f/d` tests pass on our system - just this one fails.

The test does appear to be compiled with:
```
$(eval $(call compile_template,rv64mi,-march=rv64g -mabi=lp64))
```

Thanks so much!

aswaterman, 2024-04-10:
Both GCC and Clang define the `__riscv_flen` macro if compiling with floating-point:

```$ echo | riscv64-unknown-elf-gcc -march=rv32i -mabi=ilp32 -dM -E - | grep riscv_flen
$ echo | riscv64-unknown-elf-gcc -march=rv32if -mabi=ilp32 -dM -E - | grep riscv_flen
#define __riscv_flen 32
$ echo | riscv64-unknown-elf-clang -march=rv32i -mabi=ilp32 -dM -E - | grep riscv_flen
$ echo | riscv64-unknown-elf-clang -march=rv32if -mabi=ilp32 -dM -E - | grep riscv_flen
#define __riscv_flen 32
```

So what you're seeing is a mystery to me.  I'd double-check the command line that the makefile actually generates to compile the test.

emturner, 2024-04-11:
apologies, the problem does indeed lie elsewhere. Sorry for the noise

aswaterman, 2024-04-11:
Glad you figured it out.