riscv-software-src/riscv-tests #551: TriggerDmode case fail，when accessing 8-9 trigger

zhajio1988, 2024-04-22:
our riscv core vendor specify that  tselect.index 0-7 can only be mcontrol, 8-9 can only be one of icount/ietrigger.
TriggerDmode case trigger.S traverses the index from 0-9, and the type selected is 2, at this time, icount and itrigger, etrigger can only be accessed when the type is equal to 3,4,5, which leads to an error at tselect.index 8.  How to deal with the situation?

en-sc, 2024-07-01:
> icount and itrigger, etrigger can only be accessed when the type is equal to 3,4,5, which leads to an error at tselect.index 8.

Do you mean writing `tdata1` with `tdata1.type == 2`  while `tselect` is 8 causes an illegal instruction exception on your HW?
This behavior is prohibited by the spec, namely [Chapter 5. Sdtrig (ISA Extension)] (The RISC-V Debug
Specification, Version 1.0.0-rc3):
> M-Mode and Debug Mode accesses to trigger CSRs that are used by any of the implemented triggers must succeed, regardless of the current type of the currently selected trigger.

However, the implementation of `write_store_trigger` has some flaws, e.g. `write_triggers` should disable the trigger by writing `tdata1 = 0` before writing `tdata2` (see [5.7. Trigger Module Registers]).
https://github.com/riscv-software-src/riscv-tests/blob/b0eb63a6cb11842c6aeeb7f92340d09728c35caa/debug/programs/trigger.S#L66-L78

zhajio1988, 2024-07-02:
@en-sc 
Do you mean writing tdata1 with tdata1.type == 2 while tselect is 8 causes an illegal instruction exception on your HW?
_Yes, it is._