riscv-software-src/riscv-tests #560: dhrystone.c:20:1: error: return type defaults to 'int' [-Wimplicit-int]

omerguzelelectronicguy, 2024-06-05:
I switched my compiler from `riscv64-unknown-elf-gcc` version from 12 to 14. However, it causes this error`dhrystone.c:20:1: error: return type defaults to 'int' [-Wimplicit-int]`, when I enter `make benchmark` in the main folder. I am at [this commit](8f9c68e6) in riscv-tests repo. Detailed log is below:
```
/diskd/github/riscv-tests/benchmarks/dhrystone/dhrystone.c:20:1: error: return type defaults to 'int' [-Wimplicit-int]
   20 | Proc_6 (Enum_Val_Par, Enum_Ref_Par)
      | ^~~~~~
/diskd/github/riscv-tests/benchmarks/dhrystone/dhrystone.c: In function 'Proc_6':
/diskd/github/riscv-tests/benchmarks/dhrystone/dhrystone.c:29:9: error: implicit declaration of function 'Func_3' [-Wimplicit-function-declaration]
   29 |   if (! Func_3 (Enum_Val_Par))
      |         ^~~~~~
/diskd/github/riscv-tests/benchmarks/dhrystone/dhrystone.c: At top level:
/diskd/github/riscv-tests/benchmarks/dhrystone/dhrystone.c:54:1: error: return type defaults to 'int' [-Wimplicit-int]
   54 | Proc_7 (Int_1_Par_Val, Int_2_Par_Val, Int_Par_Ref)
      | ^~~~~~
/diskd/github/riscv-tests/benchmarks/dhrystone/dhrystone.c:74:1: error: return type defaults to 'int' [-Wimplicit-int]
   74 | Proc_8 (Arr_1_Par_Ref, Arr_2_Par_Ref, Int_1_Par_Val, Int_2_Par_Val)
      | ^~~~~~
/diskd/github/riscv-tests/benchmarks/dhrystone/dhrystone_main.c: In function 'main':
/diskd/github/riscv-tests/benchmarks/dhrystone/dhrystone_main.c:120:7: error: implicit declaration of function 'Proc_5' [-Wimplicit-function-declaration]
  120 |       Proc_5();
      |       ^~~~~~
/diskd/github/riscv-tests/benchmarks/dhrystone/dhrystone_main.c:121:7: error: implicit declaration of function 'Proc_4' [-Wimplicit-function-declaration]
  121 |       Proc_4();
      |       ^~~~~~
/diskd/github/riscv-tests/benchmarks/dhrystone/dhrystone_main.c:127:21: error: implicit declaration of function 'Func_2'; did you mean 'Func_1'? [-Wimplicit-function-declaration]
  127 |       Bool_Glob = ! Func_2 (Str_1_Loc, Str_2_Loc);
      |                     ^~~~~~
      |                     Func_1
/diskd/github/riscv-tests/benchmarks/dhrystone/dhrystone_main.c:133:9: error: implicit declaration of function 'Proc_7' [-Wimplicit-function-declaration]
  133 |         Proc_7 (Int_1_Loc, Int_2_Loc, &Int_3_Loc);
      |         ^~~~~~
/diskd/github/riscv-tests/benchmarks/dhrystone/dhrystone_main.c:138:7: error: implicit declaration of function 'Proc_8' [-Wimplicit-function-declaration]
  138 |       Proc_8 (Arr_1_Glob, Arr_2_Glob, Int_1_Loc, Int_3_Loc);
      |       ^~~~~~
/diskd/github/riscv-tests/benchmarks/dhrystone/dhrystone_main.c:140:7: error: implicit declaration of function 'Proc_1' [-Wimplicit-function-declaration]
  140 |       Proc_1 (Ptr_Glob);
      |       ^~~~~~
/diskd/github/riscv-tests/benchmarks/dhrystone/dhrystone_main.c:147:11: error: implicit declaration of function 'Proc_6' [-Wimplicit-function-declaration]
  147 |           Proc_6 (Ident_1, &Enum_Loc);
      |           ^~~~~~
/diskd/github/riscv-tests/benchmarks/dhrystone/dhrystone_main.c:158:7: error: implicit declaration of function 'Proc_2' [-Wimplicit-function-declaration]
  158 |       Proc_2 (&Int_1_Loc);
      |       ^~~~~~
/diskd/github/riscv-tests/benchmarks/dhrystone/dhrystone_main.c: At top level:
/diskd/github/riscv-tests/benchmarks/dhrystone/dhrystone_main.c:242:1: error: return type defaults to 'int' [-Wimplicit-int]
  242 | Proc_1 (Ptr_Val_Par)
      | ^~~~~~
/diskd/github/riscv-tests/benchmarks/dhrystone/dhrystone_main.c: In function 'Proc_1':
/diskd/github/riscv-tests/benchmarks/dhrystone/dhrystone_main.c:258:3: error: implicit declaration of function 'Proc_3'; did you mean 'Proc_1'? [-Wimplicit-function-declaration]
  258 |   Proc_3 (&Next_Record->Ptr_Comp);
      |   ^~~~~~
      |   Proc_1
/diskd/github/riscv-tests/benchmarks/dhrystone/dhrystone_main.c: At top level:
/diskd/github/riscv-tests/benchmarks/dhrystone/dhrystone_main.c:276:1: error: return type defaults to 'int' [-Wimplicit-int]
  276 | Proc_2 (Int_Par_Ref)
      | ^~~~~~
/diskd/github/riscv-tests/benchmarks/dhrystone/dhrystone_main.c:299:1: error: return type defaults to 'int' [-Wimplicit-int]
  299 | Proc_3 (Ptr_Ref_Par)
      | ^~~~~~
/diskd/github/riscv-tests/benchmarks/dhrystone/dhrystone_main.c:314:1: error: return type defaults to 'int' [-Wimplicit-int]
  314 | Proc_4 () /* without parameters */
      | ^~~~~~
/diskd/github/riscv-tests/benchmarks/dhrystone/dhrystone_main.c:326:1: error: return type defaults to 'int' [-Wimplicit-int]
  326 | Proc_5 () /* without parameters */
```

pdonahue-ventana, 2024-06-05:
I ran into this same problem.  I didn't have time to investigate so I just stayed on the old compiler, though I'd prefer to be able to use the latest.

omerguzelelectronicguy, 2024-06-06:
I tried to supress them however when I supress them and compile, it ran differently than the compiled one with gcc12.
The one with gcc14 is stucked when I ran in spike.

omerguzelelectronicguy, 2024-06-07:
@pdonahue-ventana I found the problem. The newer compiler is better to vectorize the code as far as I understand. In both cases the march option was `gcv` but the older one doesn't produce a vector and I didn't realize the vector extension was in the march option. I also now see that I didn't supply enough information while asking this question. I am sorry for that.

closed 2024-06-07 by omerguzelelectronicguy

TommyMurphyTM1234, 2024-10-28:
Any use?

- https://github.com/riscv-software-src/riscv-tests/issues/584