riscv-software-src/riscv-tests #564: Debug: Spike `--dm-auth` causes unnecessary/distracting authentication warnings/errors

TommyMurphyTM1234, 2024-06-10:
When testing against a simulartor the `--dm-auth` option is used:

- https://github.com/riscv-software-src/riscv-isa-sim/blob/00dfa28cd71326a9b553052bf0160cb76f0e7e07/spike_main/spike.cc#L77

having been passed from `testlib.py`:

- https://github.com/riscv-software-src/riscv-tests/blob/8f9c68e6d87fa5e058c44bbdb18b129443f3119a/debug/testlib.py#L138

When using Spike, this leads to distracting warnings/error messages:
```
Error: 126 12 riscv-013.c:836 dmstatus_read_timeout(): Debugger is not authenticated to target Debug Module. (dmstatus=0x4c0c02). Use `riscv authdata_read` and `riscv authdata_write` commands to authenticate.
Error: 127 12 target.c:681 target_examine_one(): [riscv.cpu] Examination failed
Debug: 128 12 target.c:682 target_examine_one(): [riscv.cpu] examine() returned error code -4
Debug: 129 12 target.c:1780 target_call_event_callbacks(): target event 20 (examine-fail) for core riscv.cpu
Warn : 130 12 target.c:739 target_examine(): target riscv.cpu examination failed
Debug: 131 12 openocd.c:152 handle_init_command(): target examination failed
```
I see that this option may also be passed when VCS simulation is used but I am not in a position to exercise this use case to test any changes:

- https://github.com/riscv-software-src/riscv-tests/blob/8f9c68e6d87fa5e058c44bbdb18b129443f3119a/debug/testlib.py#L121-L138
- https://github.com/riscv-software-src/riscv-tests/blob/8f9c68e6d87fa5e058c44bbdb18b129443f3119a/debug/testlib.py#L239-L249

Does anybody know if it would be OK to remove the `--dm-auth` option to eliminate the distracting warning/error messages?

If it would be then I will prepare a PR.

en-sc, 2024-06-10:
> Does anybody know if it would be OK to remove the `--dm-auth` option to eliminate the distracting warning/error messages?

I would be strongly opposed to this change.

The thing is, DM authentication is a peculiar mechanism that requires running commands on a non-examined target (after an examination fails / before the examination was attempted).

An alternative solution would be to create a separate test targeting DM authentication. However, this would require a separate Spike-based target, since Spike configuration is directed by the target, not the test itself.

IMHO, just dropping the `--dm-auth` will be bad for the coverage of the testsuite and I believe these undoubtedly annoying and confusing messages should be left as-is.

TommyMurphyTM1234, 2024-06-10:
> IMHO, just dropping the `--dm-auth` will be bad for the coverage of the testsuite and I believe these undoubtedly annoying and confusing messages should be left as-is.

How so? I removed the `--dm-auth` option and ran the tests against the 4 Spike targets and the results were exactly the same as before removing the option albeit, without the annoying/distracting error/warning messages. 
```
./gdbserver.py ./targets/RISC-V/spike64.py --isolate --print-failures --remotelogfile-enable --server_cmd "openocd -d3" 2>&1 | tee test-spike64.log
./gdbserver.py ./targets/RISC-V/spike64-2.py --isolate --print-failures --remotelogfile-enable --server_cmd "openocd -d3" 2>&1 | tee test-spike64-2.log
./gdbserver.py ./targets/RISC-V/spike32.py --isolate --print-failures --remotelogfile-enable --server_cmd "openocd -d3" 2>&1 | tee test-spike32.log
./gdbserver.py ./targets/RISC-V/spike32-2.py --isolate --print-failures --remotelogfile-enable --server_cmd "openocd -d3" 2>&1 | tee test-spike32-2.log
```
What is the rationale for passing the option when authentication isn't actually used for Spike?

TommyMurphyTM1234, 2024-06-10:
OK - I think I get it now.

`testlib.py` invokes Spike with `--dm-auth` declaring that a debug connection needs to authenticate.
The `spike-*.cfg` OpenOCD scripts authenticate using `riscv authdata_read/write`.

If `testlib.py` is modified to NOT pass `--dm-auth` but the scripts are retained unchanged and so still authenticate then that's fine.
But if `--dm-auth` is passed but the OpenOCD scripts do NOT authenticate then you get this:
```
Error: 122 12 riscv-013.c:836 dmstatus_read_timeout(): Debugger is not authenticated to target Debug Module. (dmstatus=0x4c0c02). Use `riscv authdata_read` and `riscv authdata_write` commands to authenticate.
Error: 123 12 target.c:681 target_examine_one(): [riscv.cpu] Examination failed
Debug: 124 12 target.c:682 target_examine_one(): [riscv.cpu] examine() returned error code -4
Debug: 125 12 target.c:1780 target_call_event_callbacks(): target event 20 (examine-fail) for core riscv.cpu
Warn : 126 12 target.c:739 target_examine(): target riscv.cpu examination failed
Debug: 127 12 openocd.c:152 handle_init_command(): target examination failed
```
So, launching Spike with `--dm-auth` and performing authentication via the OpenOCD scripts is deliberate and is doing a basic test of the autnentication mechanism.

That was not originally obvious to me.

Closing this issue as invalid.

closed 2024-06-10 by TommyMurphyTM1234