riscv-software-src/riscv-tests #571: rv32mi-p-breakpoint fails without debugger support

danlehmann, 2024-07-10:
According to the most recent debug spec:
"Write 0 to tselect. If this results in an illegal instruction exception, then there are no triggers implemented."

However, in the breakpoint test tselect is expected to exist:
https://github.com/riscv-software-src/riscv-tests/blob/master/isa/rv64mi/breakpoint.S#L33

Notice how tcontrol right before handles illegal instruction, but tselect doesn't. So an implementation that traps will fail this test.

aswaterman, 2024-07-10:
Many of the tests in this repo are only meant to be executed on implementations that support the feature being tested. For example, the floating-point tests can’t be executed on implementations without the F extension.

danlehmann, 2024-07-10:
Thanks for the quick response. Isn't this piece of code designed to detect the existence though? It just seems to me like it uses an older version of the spec.

```
  # Skip tselect if hard-wired.
  csrw tselect, x0
  csrr a1, tselect
  bne x0, a1, pass
```

aswaterman, 2024-07-10:
I don't think the version of the spec is the question.  If the test is meant to execute only on implementations that have the `tselect` register, then the test is fine as-is.  If the test is meant to execute on any RISC-V M-mode implementation, then the test is deficient for reasons you mention.  We can resolve this one either way, but my recommendation is to simply not run the test if `tselect` doesn't exist.

danlehmann, 2024-07-10:
Got it, thanks for your help!

closed 2024-07-10 by danlehmann