riscv-software-src/riscv-tests #577: csr registers "sptbr" "sbadaddr" "mbadaddr" can't be recognized by self-compiled gcc

HUJIYONG, 2024-08-14:
My gcc was configured as:
```
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/u/workspace/riscv/toolchain/libexec/gcc/riscv64-unknown-elf/13.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/u/workspace/riscv/tools/riscv-gnu-toolchain/build/../gcc/configure --target=riscv64-unknown-elf --prefix=/home/u/workspace/riscv/toolchain --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion=gc891d8dc23e --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/u/workspace/riscv/toolchain/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=../../gcc --disable-multilib --with-abi=lp64d --with-arch=rv64gcv --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 13.2.0 (gc891d8dc23e) 
```

In general, I use option "--with-arch=rv64gcv" and other options as default.

Then I compile riscv-tests of the newest version:
```
git clone https://github.com/riscv-software-src/riscv-tests
autoupdate
autoconf
./configure --prefix=$RISCV/target
make
```
I got below errors:
```
rv32si/../rv64si/ma_fetch.S: Assembler messages:
rv32si/../rv64si/ma_fetch.S:209: Error: unknown CSR `sbadaddr'
make[1]: *** [/home/u/workspace/test/riscv-tests/isa/Makefile:101: rv32si-p-ma_fetch] Error 1
make[1]: *** Waiting for unfinished jobs....
rv32si/../rv64si/dirty.S: Assembler messages:
rv32si/../rv64si/dirty.S:25: Error: unknown CSR `sptbr'
make[1]: *** [/home/u/workspace/test/riscv-tests/isa/Makefile:101: rv32si-p-dirty] Error 1
rv32mi/../rv64mi/ma_addr.S: Assembler messages:
rv32mi/../rv64mi/ma_addr.S:106: Error: unknown CSR `mbadaddr'
make[1]: *** [/home/u/workspace/test/riscv-tests/isa/Makefile:102: rv32mi-p-ma_addr] Error 1
rv32mi/../rv64mi/illegal.S: Assembler messages:
rv32mi/../rv64mi/illegal.S:77: Error: unknown CSR `sptbr'
rv32mi/../rv64mi/illegal.S:87: Error: unknown CSR `sptbr'
rv32mi/../rv64mi/illegal.S:123: Error: unknown CSR `sptbr'
rv32mi/../rv64mi/illegal.S:159: Error: unknown CSR `mbadaddr'
make[1]: *** [/home/u/workspace/test/riscv-tests/isa/Makefile:102: rv32mi-p-illegal] Error 1
rv32mi/../rv64si/ma_fetch.S: Assembler messages:
rv32mi/../rv64si/ma_fetch.S:209: Error: unknown CSR `mbadaddr'
```
In general, some csr registers can't be recognized "sptbr" "sbadaddr" "mbadaddr"
I search in the riscv privilege spec, I got:
(1) sptbr:
I found it does exist in spec of 2016, but in 2021 or 2024: 
> The machine-mode base-and-bounds translation and protection schemes have been removed
from the specification as part of moving the virtual memory configuration to sptbr (now satp).
Some of the motivation for the base and bound schemes are now covered by the PMP registers, but
space remains available in mstatus to add these back at a later date if deemed useful.
The supervisor virtual memory configuration has been moved from the mstatus register to the
sptbr register. Accordingly, the sptbr register has been renamed to satp (Supervisor Address
Translation and Protection) to reflect its broadened role.

In 2016 spec, 'sptbr' is csr 0x180, and in 2021 spec, csr 0x180 is satp. 

(2) sbadaddr:
Still, I found this in spec of 2016, but not anything in spec 2021 or 2024.
But as (1), I find that, in 2021 spec, stval has the same index 0x243 which is same with sbadaddr. AND, they get similar description

So, I GUESS, these two csr were upgraded and renamed, but riscv-tests does't fit the new spec. I try modified the code accordingly and pass compile, but am not sure whether it would result some bugs.

Have there any body met this?

aswaterman, 2024-08-14:
Feel free to submit a PR that replaces sbadaddr with stval, apt r with satp, etc.

aswaterman, 2024-08-15:
Resolved by #578

closed 2024-08-15 by aswaterman