riscv-software-src/riscv-tests #588: bseti.S use bset rather than bseti

Squareless-XD, 2024-10-30:
In ./isa/rv64uzbs/bseti.S, bset is used in instructions rather than bseti, which triggers "Illegal Instruction" in my cross-compiler when compiling. Is this a typo in the code, or is it only caused by a certain compiler?

aswaterman, 2024-10-30:
`bset` with an immediate argument is a pseudoinstruction for `bseti`.  The test is correct as-is.  The most likely explanation is that you don't have the Zbs extension enabled.

zxcvbnm1asdf23, 2024-12-29:
> `bset` with an immediate argument is a pseudoinstruction for `bseti`. The test is correct as-is. The most likely explanation is that you don't have the Zbs extension enabled.

I also encountered the same problem. After replacing bset in bseti.s with bseti, gcc compilation passed. Is it possible that the gcc version is not the latest?

aswaterman, 2024-12-29:
> Is it possible that the gcc version is not the latest?

binutils version rather than gcc version, but yes, that is likely the explanation. (Sorry, I don’t know offhand which version is sufficient.)

closed 2026-03-18 by aswaterman