riscv-software-src/riscv-tests #610: Compiling rv32uf-fadd fails with clang due to float NaN definition

etterli, 2025-04-11:
I'm trying to build the ISA tests with PULP's LLVM riscv toolchain. I can successfully build the `rv32ui` tests but cannot compile the `rv32uf` and `rv32ud` tests.

In particular, the [testcase 11](https://github.com/riscv-software-src/riscv-tests/blob/7785754f8f2811e691ec7c1e8f3683e1ed386006/isa/rv64uf/fadd.S#L33) (testing NaN) of `rv32uf/fadd.s` cannot be built due to how the `qNaNf` value is specified in the [`test_macros.h`](https://github.com/riscv-software-src/riscv-tests/blob/7785754f8f2811e691ec7c1e8f3683e1ed386006/isa/macros/scalar/test_macros.h#L422C16-L422C25) file. The `qNaNf` value is defined as:
```
#define qNaNf 0f:7fc00000
```

Thus, the test macro (`TEST_FP_OP2_S`) expands to the following assembly code:
```
test_11:
  // some instructions
.pushsection .data;
.align 2;
test_11_data:
  .float Inf;
  .float Inf;
  .float 0.0;
  .float 0f:7fc00000;  // <-- this fails
.popsection;
```

The build now fails on the `0f:7fc00000` floating point literal. The `0f:` is marked as unexpected token. The same problem applies to the rv32-ud tests except that `0d:` instead of `0f:` is used.

I tried without success:
- Using `0x` or `0h` as prefix but this resulted in an invalid floating point literal error. It seems only actual floating point numbers work.
- Using `NaN` instead of the special `qNaNf` value. This compiled but I'm not certain whether the actual NaN value is correct and/or how I can differentiate between the qNaN or sNaN value?

An alternative solution I found and works is:
- Change the `TEST_FP_OP2_S` macro such that `.float` literal is replaced with a `.quad` literal and replace the `qNaNf` definition with the regular hex number. The generated code then is:
```
#define qNaNf 0x7fc00000

test_11:
  // some instructions
.pushsection .data;
.align 2;
test_11_data:
  .float Inf;
  .float Inf;
  .float 0.0;
  .quad 0x7fc00000;  // <-- this works
.popsection;
```

Despite this works, this solution is not very nice as I obviously cannot use the original upstream code anymore. Using a gcc toolchain to build the tests is a valid idea but I don't want to do this as we then have to setup/maintain two toolchains.

- Is there a simpler solution such that LLVM supports the GCC float literal?
- Is it planned to support LLVM in future?

aswaterman, 2025-04-15:
It took a while to find a clean workaround, but I was able to do so.

etterli, 2025-04-15:
Thanks a lot! The tests which use qNaN / cNaN now work.

However, the `sNaN` (in float and double [`fcmp`](https://github.com/riscv-software-src/riscv-tests/blob/8b9bd7b8d2df5e126351ff5cec51915bde1e9e5f/isa/rv64uf/fcmp.S#L31) & [`fmin`](https://github.com/riscv-software-src/riscv-tests/blob/8b9bd7b8d2df5e126351ff5cec51915bde1e9e5f/isa/rv64uf/fmin.S#L35)) is undefined for my clang.
```
riscv-tests/isa/rv32uf/../rv64uf/fcmp.S:31:244: error: invalid floating point literal in '.float' directive
  test_10: li gp, 10; ... test_10_data: .float SNaN; .float 0; .float 0.0; .word 0; .popsection
                                               ^
```

Did you test with clang and with which version? Maybe my 15.0.0 based is missing it. I also could not find a definition somewhere for a more recent version. But my research lead me to the following statement in [LLVM langref](https://llvm.org/docs/LangRef.html#behavior-of-floating-point-nan-values):
"Floating-point math operations are allowed to treat all NaNs as if they were quiet NaNs. For example, “pow(1.0, SNaN)” may be simplified to 1.0."

As far as I understand, the LLVM relies on the floating point environment to handle signaling NaNs by checking the corresponding flags. Therefore, I don't think the SNaN symbol is defined and handling it with the hex number as it was initially is probably the cleanest way. Should I create a proposal for SNaN macros similar to the one you created?

aswaterman, 2025-04-15:
I didn't test with LLVM at all.  I was just hoping that this was a case where LLVM would have emulated whatever Binutils did.  Evidently not.  Looks like this one will need to be handled similarly to how I handled qNaN.

aswaterman, 2025-04-15:
Found a better workaround; will make a PR soon.

aswaterman, 2025-04-16:
OK, see #613, and please close this issue again if it resolves the Clang build.

etterli, 2025-04-16:
It works. Thank you for your support :)

closed 2025-04-16 by etterli

aswaterman, 2025-04-16:
Cool!

PR #612 (2025-04-15): Remove use of qNaN/sNaN to fix LLVM build
Resolves #610